Bodmer/JPEGDecoder #2: I get the list of error showen under

fredericplante, 2016-11-17:
> Arduino: 1.6.5 (Windows XP), Board: "Arduino Due (Programming Port)"

Build options changed, rebuilding all

WARNING: library TFT_HX8357 claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [sam] architecture(s).

TFT_SDcard_jpg_v1.ino: In function 'void loop()':
TFT_SDcard_jpg_v1:114: error: 'drawSdJpeg' was not declared in this scope
TFT_SDcard_jpg_v1:135: error: 'createArray' was not declared in this scope
TFT_SDcard_jpg_v1.ino: In function 'void drawSdJpeg(char*, int, int)':
TFT_SDcard_jpg_v1:149: error: 'renderJPEG' was not declared in this scope
TFT_SDcard_jpg_v1.ino: In function 'void renderJPEG(int, int)':
TFT_SDcard_jpg_v1:194: error: 'showTime' was not declared in this scope
TFT_SDcard_jpg_v1.ino: In function 'void createArray(const char*)':
TFT_SDcard_jpg_v1.ino:258:40: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
Multiple libraries were found for "SPI.h"

 Used: C:\Documents and Settings\So sory\Application Data\Arduino15\packages\arduino\hardware\sam\1.6.9\libraries\SPI

 Not used: C:\Documents and Settings\So sory\My Documents\Arduino\libraries\SmingCore

'drawSdJpeg' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Any cue to give me?

Bodmer, 2016-11-23:
IDE 1.6.5 has an issue with functions in the sketch that are not declared before setup(). Use the latest IDE (1.6.12) or move the drawSdJpeg function to before setup().

closed 2016-11-23 by Bodmer

fredericplante, 2016-11-23:
I have installed the 1.6.12 last night, so far I make the SerialCsvOut example work on an ATmega2506 work, by changing the nature of the _uint8 *pImg;_ to _unsign int *pImg_, the code compile, I have not tested on board yet. Some how, when I fix this line, it also work on the 1.6.5 version of the IDE, but the nature of the variable has to be _short unsign int *pImg_ So great I will keep using 1.6.5 ;)

As for the 2 TFT examples, bouette, I get error message all over, it totally crash for many reason. So since your library give out RGB 24bit pixel by pixel, I'm going to try to merge this and the UTFT library, since it covers most TFT and which has a draw bitmap function, by filling the memory of a full line.
 
**320 pixels * 3 = 960 bytes** 

Make it draw, one line at the time. I will give you news about it

Bodmer, 2016-11-23:
Note that the original library has been modified to return a 16 bit colors in 565 format.

fredericplante, 2016-11-23:
Oh! This is important to note. Mk thank you for the extra details :)

Bodmer, 2016-12-09:
OK, also note that the 16 bit colour format needed by UTFT might be RGB or BGR (5 bits of red and blue swapped)

fredericplante, 2016-12-09:
Yeap! All is fine, your library work fine, my arduino folder was a mess, all is good now. ;)