Bodmer/JPEGDecoder #5: Please add library.json and library.properties files to your repo

squix78, 2017-01-21:
Can you please add library.json and library.properties files to your repository? This allows them to be available through the library manager in the arduino IDE and the platform io lib manager.

See this documentations:
http://docs.platformio.org/en/latest/librarymanager/config.html
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

Once you have added the files the platformio and arduino ide teams need to be asked to add your repository to their scanning job...

And here two examples:
https://github.com/squix78/esp8266-oled-ssd1306/blob/master/library.json
https://github.com/squix78/esp8266-oled-ssd1306/blob/master/library.properties

Bodmer, 2017-01-22:
OK,  will do, thanks.

Bodmer, 2017-01-23:
OK, I have added the files.

Thanks.

squix78, 2017-01-23:
Thanks a lot! I think for the library managers to pick up the library files you'll also have to do a release here on github...

Bodmer, 2017-01-24:
OK, thanks again.

closed 2017-01-24 by Bodmer