Bodmer/JPEGDecoder #7: Some help...

rwijbenga, 2017-02-06:
I am trying to get your LIB to work with the PlaneSpotter Collor sketch. I saw on the PlaneSpotter page you had made some improvements. 

Could you help me out on what to change in the PlaneSpotter sketch to make it work ?

My programming skills are not that good yet.

I now see the plane's, but the stay on the screen, and the map is missing.

The loading and refresh are better with you plugin, I can see that already.. :)

Bodmer, 2017-02-07:
[Try the version here](https://github.com/Bodmer/esp8266-plane-spotter-color).  You will need the JPEGDEcoder library and the [TFT_ILI9341_ESP library here.](https://github.com/Bodmer/TFT_ILI9341_ESP)

NodeMCU connections to TFT display [as detailed here](https://github.com/Bodmer/TFT_ILI9341_ESP/blob/master/User_Setup.h#L10-L25).

rwijbenga, 2017-02-07:
Thank you! 

It works super now!

Are you planning on supporting the touch panel?

I am going to make a laser cut housing for the display and the ESP. Maybe I can send you one also, for the help..

Remco

Bodmer, 2017-02-07:
OK, great to hear it is working for you.

I have a few other projects on the go so adding touch is not a priority, but I will keep an eye on the Squix developments on the project and port across any touch handling if it is added.

Thanks for the offer of a case, but your thank you is enough!   I have a 3D printer and so that would be my route to make a case.

closed 2017-02-07 by Bodmer