Bodmer/JPEGDecoder #10: Help with implementation on Photon (Particle)

fredvaz, 2017-04-22:
Hello, 

i want use this library on Photon. It's like arduino programation. I already compiled without erros, but when i print to serial the buffer of the decoded image (Baboon), and i run imshow(baboon_decoded) on Matlab, i don't saw the real image of baboon... 

It's possible that math operation go wrong on Photon? It's a ARM M3, 120Mhz, 128kb SRAM.
[image]

Thanks for your attention

Bodmer, 2017-04-23:
It looks like you are assuming the output is a raster scan image. Actuallly the output is a set of image tiles, typically blocks of 16x16 pixels. These tiles are then plotted in raster format. This is described in the jpeg documentation.
Also be aware that each pixel is 16 bits in 565 colour format.
The image output suggests the content of a tile buffer is being repeated, so check you code adaptions.
I don't have a Photon, so can't help further.

closed 2017-04-23 by Bodmer

fredvaz, 2017-04-23:
First at all i am gratefull for your help! 

I changed comps to 3, for RGB correctly? (it's the only thing a changed on your code)

Ok, on Matlab i maked a cast to uint8 -.-" i must have uint16. Now i am use a Arduino Mega, and the result is this:
[image]

It's better! ahaha But i have to improve... 
 My Matlab code is:

`%% Converto to Array
n = 0;
for y=1:16:300
    for x=1:16:300
        % Raster mod
        for i=y:y+15
            for j=x:x+15
                n = n + 1;
                for k=1:3
                    rgb(i,j,k) = uint16(buffer2(n,k));
                end
            end
        end
    end
end`