Bodmer/JPEGDecoder #12: JPEGDecoder crashing ESP8266

TSJim, 2017-06-14:
Hi Bodmer...

I'm using your JPEGDecoder in a sketch, and you were nice enough to fix the JPEG artifact bug I was encountering, so here is another issue I hope you can help with.

When I run the JPEGDecoder in a stand-alone sketch, it works fine.  But when I run it in my slightly more complex sketch, it resets the ESP8266.

I think I have isolated the problem to the jpegRender function (if I comment that out, no resets occur, but of course no image is displayed either).

Things I have tried:

  o  Running on just a ESP8266 dev board (Huzzah Feather and NodeMCU) and on a plain ESP8266-12F in my circuit (both are reset by the JPEGDecoder functions)

  o  Using different JPEG images (including your samples) and even very small JPEG images

Can I please give you my sketch for you to try and see if you can figure out why JPEGDecode is causing a reset?

Thank you!

Bodmer, 2017-06-17:
The ESP8266 has an internal watchdog that may be resetting the processor. Try adding a yield(); statement before calling the jpegRender function.

TSJim, 2017-06-26:
Hi Bodmer...

Yes, I have been using yield() with ESPs for a long while!  ;)  Just a part of life.  I have them in this sketch as well.  But I appreciate the suggestion.

Today I replaced your **drawJpeg()** function that I was using in my sketch with your lesser-known **drawRaw()** function from your older TFT_ILI9341_ESP stuff, and it works great!  No crashes, and seems just as fast as the JPEG stuff, although I didn't time it.  I put the 565 raw data file in the SPIFFS, and everything is awesome!  I didn't even need to include the TFT_ILI9341_ESP library because I guess the stuff that drawRaw() needs is also in your "TFT_eSPI" library that I'm using because of the string printing stuff like centered strings.

I think I might do an Instructable about displaying raw 565 files from SPIFFS  using the newer TFT_eSPI library because it is not something that people seem to know about very much.  It seems like everyone thinks you can only read JPEG files from SPIFFS.  I think raw is be a great solution for a lot of projects.

However, this does make me wonder why the **drawJpeg()** function crashes my same sketch, but the drawRaw() does not.  I'm guessing it is using more memory to do the JPEG decoding, or something.  But it is way beyond my abilities to track that down.

Thank you!  You're awesome, as everyone knows.

Bodmer, 2017-07-03:
OK, this might be the same problem as the #13 issue, where the struct that is generated by the decoder from the jpeg file returns duff values.  I am guessing this is a ESP8266 core bug as I see no reason this to happen.

closed 2017-10-09 by Bodmer