Bodmer/JPEGDecoder #13: ESP8266: JPG image from SPIFFS gets wrong decoding

nikola-winmaker, 2017-06-18:
Hi all,
Great library! you made a great job!

I'm having some issue when running JPG decoding from SPIFFS,
The same sketch running from SD card runs perfectly. 
The same JPG image is on SD card and on spiffs.

What i'm seeing when calling jpegInfo() is that values are shifted down:
Width     0
Height    160
Components 128
and so on...

Picture is 160x128.

This is only happening when running sketch from SPIFFS.
The image is uploaded to SPIFFS via Arduino  ESP8266 Sketch Data Upload tool.
I uploaded JPG image for reference.

Best Regards,
Nikola

[image]

Bodmer, 2017-06-21:
This is not something I have encountered.  I suspect this is a bug in the SPIFFS code that is part of the ESP8266 board package core.  [See here](https://github.com/esp8266/Arduino/pull/3261). In which case this should be fixed when version 2.4.0 of the Arduino core is released.

Bodmer, 2017-06-23:
OK, update: I have witnessed a similar problem where the values returned by jpegInfo() are wrong. This appears to be due to a corrupted pointer that affects the values read out of a struct that contains the values.  One of the values (Components) is a hard coded value of 1 and this seems to end up being reported as a different value.

I think this might be a bug in the ESP8266 core but have not been able to isolate the root cause.

closed 2018-01-23 by Bodmer