Bodmer/JPEGDecoder #15: Image starts decoding fine, then gets stuck repeating same pixel color

zchester, 2017-09-14:
I am working on a project and using Jans jpeg decoding on arduino tutorial as a guide. I used his code for the sending, and receiving and your code for decoding. I am not using the arduino camera though. I am attempting to just simply send a picture that I took using my computer. It is a jpg file, I reduced the size to 640x480, and it starts sending and populating the image one pixel at a time, but once it gets about 25% done, it stops sending color pixels and just starts just populating with a repeated single pixel color boxes. BUT.....when I attempt to send an image that is in the example folder (the tiger.jpg) it works perfectly fine. The whole image gets recieved. Do you have any ideas to help me out??? I confirmed the image was not a progressive jpeg.

Bodmer, 2017-09-27:
It looks like the bytes are not getting piped from the image source into the decoder correctly. The decoder uses a callback mechanism to request blocks of bytes but it looks like this callback is not getting the bytes it needs, so it aborts and you just get back the last block.

To enable me to understand what is happening in your setup I will need to see some example code that I can use to demonstrate the problem.

zchester, 2017-09-27:
Here is the code and the image that I used. 

[image]

[Original_jpeg2.zip](https://github.com/Bodmer/JPEGDecoder/files/1338770/Original_jpeg2.zip)

zchester, 2017-10-08:
Did you find any issues?

Bodmer, 2017-10-09:
I have tried your image on a Mega board and as you say it gets stuck repeating the came blank block of colour.  I suspect this is down to an "int" type variable in the decoder which overrflows on this particular image and perhaps needs to be more than 16 bits.

I have tested the same image on an ESP8266 with the JPEGDecoder library and the image decodes fine, so this seems to support my suspicion.  I also suspect the problem variable(s) is/are in the picojpeg.c file which has been written by Rich Geldreich. I will see if I can isolate the problem.

Bodmer, 2017-10-09:
The JPEGDecoder library also decodes the image correctly when using an Arduino Due.

I had a look at the picojpeg.c code and see the int16 tpyes are defined as a "short", so this means that 16 bit numbers are OK.

I can only conclude that the problem experienced is only manifesting itself on a Mega, perhaps the Mega runs out of RAM during the decoding process?

Bodmer, 2017-11-16:
More tests show this is only an issue on the Mega, I have no idea why :-(

closed 2017-11-16 by Bodmer