Bodmer/JPEGDecoder #16: Image starts decoding fine, then gets stuck repeating same pixel color

zchester, 2017-09-14:
I am working on a project and using Jans jpeg decoding on arduino tutorial as a guide. I used his code for the sending, and receiving and your code for decoding. I am not using the arduino camera though. I am attempting to just simply send a picture that I took using my computer. It is a jpg file, I reduced the size to 640x480, and it starts sending and populating the image one pixel at a time, but once it gets about 25% done, it stops sending color pixels and just starts just populating with a repeated single pixel color boxes. BUT.....when I attempt to send an image that is in the example folder (the tiger.jpg) it works perfectly fine. The whole image gets recieved. Do you have any ideas to help me out??? I confirmed the image was not a progressive jpeg.

zchester, 2017-09-14:
In the comment you can read the conversation between myself (Zech) and Jan. If that helps. I have images attached there. http://www.deviceplus.com/how-tos/arduino-guide/jpeg-decoding-on-arduino-tutorial/

zchester, 2017-09-22:
Hey where you able to repeat the issue that I am having?
Thanks

Bodmer, 2017-09-27:
I've been off traveling hence my tardiness in responding..

See my response to Issue #15

closed 2017-10-09 by Bodmer