Bodmer/JPEGDecoder #17: ESP32 does now support SPIFFS

universam1, 2017-09-20:
https://github.com/Bodmer/JPEGDecoder/blob/4164a392bd7ebde7692e08c6e8020b6499c1b921/src/JPEGDecoder.h#L39

Is this still correct?
https://github.com/espressif/arduino-esp32/tree/master/libraries/SPIFFS

Bodmer, 2017-09-27:
Thanks for letting me know about the SPIFFS support for the ESP32. I will see if I can get it working on an ESP32.

Ierlandfan, 2017-10-18:
I use the unofficial SPIFFS library for the ESP32 (https://github.com/copercini/arduino-esp32-SPIFFS) I Only thing I had to add in the unofficial one was: 

-- JPEGDecoder.cpp --

#ifdef USE_SPIFFS
    #include <SPIFFS.h> <<<<<<<<
	#include <FS.h>

Hubberthus, 2017-12-08:
The official library is already there, as @universam1 wrote it.
It has to include "SPIFFS.h" if you want to use it. Including only "FS.h" is not enough.
So the whole `#ifdef ESP32 ... #ifndef ESP32` stuff can be reduced in "JPEGDecoder.h" to this:
`#define LOAD_SPIFFS
#define FS_NO_GLOBALS
#include <FS.h>
#include <SPIFFS.h>`

Bodmer, 2018-01-23:
Support for ESP32 SPIFFS and an example for ESP32 added.

closed 2018-01-23 by Bodmer