Bodmer/JPEGDecoder #18: library compile errors

lnetwalker, 2018-03-13:
Hi all, 

esp core version 2.4.1
arduino ide 1.8.6
JPEGDecoder 1.7.8 ( and clone of todays master branch )

in my project I have an arducam and a display connected.
what i want to do is read the jpeg image from arducam, convert it and display the image.
My missing part is the conversion from jpeg to the display format. i included the jpeg library ( nothing else ,i called nothing in the lib )
and get the following error messages when compiling.
could anybody give me a hint what went wrong ?

Line 11 in my code is the include of the jpeg library
snip
```
// Demo Code for Frickelboard1
// Do in a loop
// read image from Cam and put on Display
// read BMP180 (i2c) and print on Display

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <Adafruit_BMP085.h>    // BMP180
#include <JPEGDecoder.h>        // JPEG decoder library

```
```
Arduino: 1.8.6 Hourly Build 2018/03/02 12:25 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/memory:81:0,
                 from /home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/FS.h:24,
                 from /home/DEVOLO/hartmut.eilers/ownCloud/Arduino/libraries/JPEGDecoder/src/JPEGDecoder.h:41,
                 from /home/DEVOLO/hartmut.eilers/ownCloud/Projects/HuSch/Frickelboard1/DemoCode/CamAndSensor2Display/CamAndSensor2Display.ino:11:
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:260:29: error: macro "swap" requires 3 arguments, but only 2 given
  swap(std::get<0>(_M_t), __p);
                             ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:266:27: error: macro "swap" requires 3 arguments, but only 1 given
       swap(unique_ptr& __u) noexcept
                           ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:269:21: error: macro "swap" requires 3 arguments, but only 2 given
  swap(_M_t, __u._M_t);
                     ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:447:29: error: macro "swap" requires 3 arguments, but only 2 given
  swap(std::get<0>(_M_t), __p);
                             ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:457:27: error: macro "swap" requires 3 arguments, but only 1 given
       swap(unique_ptr& __u) noexcept
                           ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:460:21: error: macro "swap" requires 3 arguments, but only 2 given
  swap(_M_t, __u._M_t);
                     ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:483:28: error: macro "swap" passed 4 arguments, but takes just 3
   unique_ptr<_Tp, _Dp>& __y) noexcept
                            ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h:484:19: error: macro "swap" requires 3 arguments, but only 1 given
     { __x.swap(__y); }
                   ^
In file included from /home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h:52:0,
                 from /home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/memory:82,
                 from /home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/FS.h:24,
                 from /home/DEVOLO/hartmut.eilers/ownCloud/Arduino/libraries/JPEGDecoder/src/JPEGDecoder.h:41,
                 from /home/DEVOLO/hartmut.eilers/ownCloud/Projects/HuSch/Frickelboard1/DemoCode/CamAndSensor2Display/CamAndSensor2Display.ino:11:
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h:852:43: error: macro "swap" requires 3 arguments, but only 1 given
    __shared_ptr(std::move(__r)).swap(*this);
                                           ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h:860:41: error: macro "swap" requires 3 arguments, but only 1 given
  __shared_ptr(std::move(__r)).swap(*this);
                                         ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h:868:43: error: macro "swap" requires 3 arguments, but only 1 given
[... 46 lines cut ...]
                   ^
In file included from /home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/memory:82:0,
                 from /home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/FS.h:24,
                 from /home/DEVOLO/hartmut.eilers/ownCloud/Arduino/libraries/JPEGDecoder/src/JPEGDecoder.h:41,
                 from /home/DEVOLO/hartmut.eilers/ownCloud/Projects/HuSch/Frickelboard1/DemoCode/CamAndSensor2Display/CamAndSensor2Display.ino:11:
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h:431:52: error: macro "swap" requires 3 arguments, but only 2 given
     swap(shared_ptr<_Tp>& __a, shared_ptr<_Tp>& __b) noexcept
                                                    ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h:432:19: error: macro "swap" requires 3 arguments, but only 1 given
     { __a.swap(__b); }
                   ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h:517:48: error: macro "swap" requires 3 arguments, but only 2 given
     swap(weak_ptr<_Tp>& __a, weak_ptr<_Tp>& __b) noexcept
                                                ^
/home/DEVOLO/hartmut.eilers/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h:518:19: error: macro "swap" requires 3 arguments, but only 1 given
     { __a.swap(__b); }
                   ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Bodmer, 2018-03-14:
The problem is that ArduCAM.h specifies a marcro called swap() at line 108 and so does the SPIFFS filing system create a macro called swap(). The SPIFFS library is called up by the JPEGDecoder library and hence the conflict is shown.

So the incompatibility is actually between the ArduCAM library and the SPIFFS library.

I suggest you rename the swap macro in the ArduCAM library files (e.g in the .h and .cpp with a search and replace) or do not use SPIFFS.

closed 2018-03-14 by Bodmer

lnetwalker, 2018-03-14:
I guess that's the problem, now everything compiles fine and I can test tomorrow on the hardware.

Thanks for your fast help and the great effort regarding JPEGDecoder! Nice work !!

Bodmer, 2018-03-14:
Great, hope your project runs smoothly now.

Some libraries also create macros for min() and max() so that can cause problems too so that is something to look out for.