Bodmer/JPEGDecoder #19: src/JPEGDecoder.cpp

nopnop2002, 2018-04-14:
Something wrong Line 348-367

#if defined (LOAD_SD_LIBRARY) || defined (LOAD_SDFAT_LIBRARY)

// Call specific to SD filing system in case leading / is used
int JPEGDecoder::decodeSdFile(const char *pFilename) {

	File pInFile = SD.open( pFilename, FILE_READ);

	return decodeSdFile(pInFile);
}

int JPEGDecoder::decodeSdFile(const String& pFilename) {
#if !defined (ARDUINO_ARCH_SAM)
	File pInFile = SD.open( pFilename, FILE_READ);

	return decodeSdFile(pInFile);
#else
	return -1;
#endif
}

nopnop2002, 2018-04-16:
If you can speek Japanese, I can speek too.

Bodmer, 2018-04-23:
I had a look and cannot see what change you made, also there is an error in your solution code.

pinFile should read pInFile?

What error message do you get with SdFat?

I do not speak Japanese other than the usual greeting!

Bodmer, 2018-06-06:
I am not sure I understand the problem.  It seems to work for me with Mega and Due.  What processor are you using? What IDE version?

nopnop2002, 2018-06-06:
I'm using STM32F103 processor.
IDE version is 1.8.5
define LOAD_SDFAT_LIBRARY.

Thank you.

anlpereira, 2018-06-19:
I'm using the ESP32 Wroon board. It is great. It takes about 300ms to draw a 240x320 Image on an ILI9341 2.8" display. 

https://www.youtube.com/watch?v=1VqzZrW_9JY

Bodmer, 2018-08-07:
This problem seems to be related to the STM32F103. I have not seen any issues with other boards.

closed 2018-08-07 by Bodmer