Bodmer/JPEGDecoder #21: JpegDec.decodeArray() always sets all the values to 0 on esp8266

emilekm2142, 2018-05-10:
Serial.println(("==============="));
			Serial.println(("JPEG image info"));
			Serial.println(("==============="));
			Serial.print(("Width      :")); Serial.println(JpegDec.width);
			Serial.print(("Height     :")); Serial.println(JpegDec.height);
			Serial.print(("Components :")); Serial.println(JpegDec.comps);
			Serial.print(("MCU / row  :")); Serial.println(JpegDec.MCUSPerRow);
			Serial.print(("MCU / col  :")); Serial.println(JpegDec.MCUSPerCol);
			Serial.print(("Scan type  :")); Serial.println(JpegDec.scanType);
			Serial.print(("MCU width  :")); Serial.println(JpegDec.MCUWidth);
			Serial.print(("MCU height :")); Serial.println(JpegDec.MCUHeight);
			Serial.println(("==============="));

always show only zeros. I am sure that the array is a correct image. I use ESP8266 on Arduino 2.3.0 and can't update to 2.4.1 due to visualMicro plugin for visual studio and using Templates that work only with ms compiler. I read that it is a bug in Arduino Core, also there is a function drawRaw() that works, but I could not find it.

Bodmer, 2018-05-11:
Does your array definition look like this:
https://github.com/Bodmer/JPEGDecoder/blob/master/examples/MCUFRIEND_kbv/jpeg_kbv/jpeg1.h#L6

emilekm2142, 2018-05-11:
The array you posted works ://
Also,  excuse me, I forgot to mention that the array is first decoded from base64.

```
std::vector<unsigned char> decodedData = base64_decode(img);
JpegDec.decodeArray(decodedData.data(), decodedData.size()); //decodedData.data() equivalent to &decodedData[0]
```

I am sure that base64_decode function gives a proper jpg.
base64_decode -> https://pastebin.com/4S9rbNUB
memchr.h -> https://pastebin.com/PBAM8NtP
You will need a standardCplusPlus by Legacy library from the manager to run that. It works on a PC though.

Bodmer, 2018-05-11:
I suspect this is an array format problem. The array is just a copy of the Jpeg file in a simple c coded byte array that is stored in program memory.