Bodmer/JPEGDecoder #22: JPEGDecoder::readSwappedBytes(): wrong colormapping when reading jpeg from SPIFF

codebeat-nl, 2018-06-04:
Hi, thanks for the nice library. I use your library in combination with the _TFT_eSPI_ library and use the example made by you _TFT_sPIFFS_jpeg.ino_. Also tried the _TFT_flash_jpg.ino_ example and operates okay however couldn't figure out how to create a valid formatted h file (no details on this) to add/change the example so I upload a jpeg file and use the SPIFFs example instead. 

However, by running the modified example, the colors seems to be swapped, red = blue and blue = red.  By changing your code at the `JPEGDecoder::readSwappedBytes()` function, I get the correct colors.  I swapped the variable names at the lines `216` and `218` in _JPEGDecoder.cpp_, blue points to red and red points to blue:
			
```
                        const uint8_t *pSrcB = image_info.m_pMCUBufR + src_ofs;
			const uint8_t *pSrcG = image_info.m_pMCUBufG + src_ofs;
			const uint8_t *pSrcR = image_info.m_pMCUBufB + src_ofs;

```

Hardware I have used:
- ESP8266 (Esp-12)
- ST7735_REDTAB

codebeat-nl, 2018-06-04:
-----------------------------

EDIT: Just post it but figured out that using T7735_REDTAB is the problem, it must be T7735_BLACKTAB in my case.  When using the other T7735_ defines I get out of alignment picture so didn't notice this color pallete change in the examples by being focussed on the alignment issue.

Bodmer, 2018-06-05:
OK! Unfortunately the TFTs vary in the colour format needed (RGB or BRG) and different processors have different endianess in 16 bit variables so these are both reasons why colours can appear corrupted and it does get confusing.

The JpegDecoder library has two functions, readSwappedBytes() and readBytes() which handle the endianess but, as you discovered, selecting the right TFT driver setup fixes the colour order.

There is a [createArray() function here](https://github.com/Bodmer/JPEGDecoder/blob/master/examples/Other%20libraries/SPIFFS_Jpeg/JPEG_functions.ino#L149) that converts SPIFFS jpeg files into array format. The function can be used as indicated [in this commented out line](https://github.com/Bodmer/JPEGDecoder/blob/master/examples/Other%20libraries/SPIFFS_Jpeg/SPIFFS_Jpeg.ino#L128).

Unfortunately I never seem to find time to write a user manual for my libraries but it is on my "To do" list!

closed 2018-06-05 by Bodmer