Bodmer/JPEGDecoder #23: Issue on My code

anlpereira, 2018-06-05:
Hi Bodmer, I don't know what is happening with my code. Some image files doesn't decompress at all. In the middle of the image it stops. See this video on youtube you will understand what I mean.

https://www.youtube.com/watch?v=RFY7xMxnhH8

Bodmer, 2018-06-05:
Which processor are you using?

Bodmer, 2018-06-06:
I am guessing you are using a Mega as that processor seems to stop decoding sometimes during the decode. See Issue #15.

The same image decodes fine on other processors so I can only assume the C code itself is OK but that for some reason the Mega gets stuck, it seems the Mega falls over when doing the maths to decode the image but I have not tracked down where this happens. The problem appears to be deep inside the picojpeg code somewhere and only manifests itself in very particular circumstances.

anlpereira, 2018-06-06:
Yes, I'm using Mega2560. I will try with ESP32 that has already arrived. Thanks

Bodmer, 2018-06-06:
OK, unfortunately I have not tracked down what is causing this, it only happens on a Mega.

closed 2018-06-06 by Bodmer