Bodmer/JPEGDecoder #25: Name of the Library is Wrong :)

prof955, 2018-07-06:
Hi,
firstly thank you for that great library.

but i believe your library name should be JPEGDecoder+TFT Shower :)
because it seem libray don't do decode Only. decode and draw images on screen too.
and if i am correct(i hope not) it cannot do Decode Only...

am i right?

please say NO :)  and point me where do i mistaken.

i found your library while i am looking library to decode PNG/BMP(or JPEG at least) image to use on my projects.  
but my project don't include any TFT screen... so i stuck. 
i searched in codes you wrote. i tried to take just decode part from JPEG_functions but no luck.
it's seems nested with tft draw functions.

i hope i am wrong(about library name) and you can prove i am wrong by showing me where the Only decoding part.

prof955, 2018-07-06:
is this part what i am looking for???

`//====================================================================================
//   Open a Jpeg file and dump it to the Serial port as a C array
//====================================================================================
void createArray(const char *filename)`

Bodmer, 2018-07-06:
The library is a decoder for the Arduino environment to support TFT display libraries. The examples show the use of the decoder with different TFT libraries. The decoder converts all colour values to 16 bits so you may find this is not what you want.

The basic decoder code is picojpeg so that is a good place to start:
https://github.com/richgel999/picojpeg

closed 2018-07-06 by Bodmer

prof955, 2018-07-06:
ok. thank you for redirecting me.
actually i take a look at that before ending up in here. 
didn't saw any use for myslef.

i believe i can use your library with some help.
i tried to find how to use again and `createArray` is serial print of all data of jpeg image.
data is in: ` while ( jpgFile.available()) { data = jpgFile.read();`
but it is just seems uint8_t  0xFF(as an example) bytes.
is it possible to decode as uint16_t 0xFFFF(as example) ?

i was just using funny language and i am sorry if it was uncomfortable / not funny for you.
was it?

prof955, 2018-07-06:
or i can ask like that:

i want to use tft.`drawRGBBitmap `so i will need RGB values. 
Can i get from your library?

its like that in GFX libray: `drawRGBBitmap(int16_t x, int16_t y, const uint16_t bitmap[], int16_t w, int16_t h)`