Bodmer/JPEGDecoder #26: Tool to convert JPG files to the PROGMEM array

sst2001, 2018-07-26:
Hello,
I could not use the existing tools online and offline to convert JPG files into the PROGMEM array.
Can you please specify which tool did you use to create the EagleEye, Baboon40, etc. and other H files you include in the example?
Thx!

Bodmer, 2018-07-26:
In the examples there is a [createArray()](https://github.com/Bodmer/JPEGDecoder/blob/master/examples/Adafruit_GFX/Huzzah_Jpeg/JPEG_functions.ino#L140) function, the calling line is commented out.

sst2001, 2018-07-27:
Thx for the prompt reply. Since I do not have working SD on my ESP32 with ST7735, is there an external tool to do it? I am using Arduino IDE on Mac.  I tried different online converters and Apps but the C array they produced is not compatible with your Lib. 
Do you have a compile external program with this procedure?
Thx a lot!

Bodmer, 2018-07-29:
The images can also be read from the ESP32 SPIFFS and decoded into an array.

I will have a look for an online tool.

Bodmer, 2018-07-29:
Most online converters change the format, but I found a simple one:

``// We need this header file to use FLASH as storage with PROGMEM directive

// The following online tool can be used to convert a jpeg to an array:
// http://tomeko.net/online_tools/file_to_hex.php?lang=en
// Copy and paste the output from the online to the array below.
#include <pgmspace.h>

const uint8_t imade[] PROGMEM = {
// paste your output here

};

sst2001, 2018-08-01:
WOW! thank you so much, this works. 

However, while it works for some files, for others it's not, and cause the  board to keep restart. 

So attached gr100 works while the design-de..... will cause constant restarts. Is there a limit I am crossing?

[image]
[image]

sst2001, 2018-08-01:
Also, when I am converting the attach the file from your example, I am getting only partial picture displayed, and the HEX array is far larger than yours. The attached file was used. 
[image]

sst2001, 2018-08-01:
I changed your program to support SD card reading on TTGO VS V1.0 (let me know if you want me to upload here, I will clean and will), I am getting a difference from your file:

Your files looks:
uint8_t EagleEye[] PROGMEM = {
0xFF,0xD8,0xFF,0xE0,0x00,0x10,0x4A,0x46,0x49,0x46,0x00,0x01,0x01,0x01,0x00,0xB4,0x00,0xB4,0x00,0x00,0xFF,0xDB,0x00,0x43,0x00,0x04,0x03,0x03,0x03,0x03,0x02,0x04,
0x03,0x03,0x03,0x04,0x04,0x04,0x05,0x06,0x0A,0x06,0x06,0x05,0x05,0x06,0x0C,0x08,0x09,0x07,0x0A,0x0E,0x0C,0x0F,0x0E,0x0E,0x0C,0x0D,0x0D,0x0F,0x11,0x16,0x13,0x0F,
0x10,0x15,0x11,0x0D,0x0D,0x13,0x1A,0x13,0x15,0x17,0x18,0x19,0x19,0x19,0x0F,0x12,0x1B,0x1D,0x1B,0x18,0x1D,0x16,0x18,0x19,0x18,0xFF,0xDB,0x00,0x43,0x01,0x04,0x04,
0x04,0x06,0x05,0x06,0x0B,0x06,0x06,0x0B,0x18,0x10,0x0D,0x10,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,
0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0x18,0xFF,0xC0,
0x00,0x11,0x08,0x00,0x80,0x00,0x80,0x03,0x01,0x22,0x00,0x02,0x11,0x01,0x03,0x11,0x01,0xFF,0xC4,0x00,0x1D,0x00,0x00,0x03,0x01,0x00,0x03,0x01,0x01,0x01,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x05,0x06,0x07,0x04,0x02,0x03,0x08,0x01,0x09,0x00,0xFF,0xC4,0x00,0x3F,0x10,0x00,0x02,0x01,0x02,0x04,0x03,0x06,0x05,0x01,0x05,0x07,
0x03,0x05,0x00,0x00,0x00,0x01,0x02,0x03,0x04,0x11,0x00,0x05,0x12,0x21,0x06,0x31,0x41,0x13,0x14,0x22,0x51,0x61,0x71,0x07,0x32,0x81,0x91,0xA1,0xB1,0x23,0x42,0xC1,
0xD1,0xF0,0x15,0x24,0x25,0x33,0x52,0x62,0x72,0x08,0x34,0x92,0x16,0x43,0x53,0xA2,0xE1,0xFF,0xC4,0x00,0x1A,0x01,0x00,0x02,0x03,0x01,0x01,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x02,0x03,0x00,0x01,0x04,0x05,0x06,0xFF,0xC4,0x00,0x28,0x11,0x00,0x02,0x02,0x01,0x04,0x01,0x04,0x01,0x05,0x01,0x00,0x00,0x00,0x00,
0x00,0x00,0x01,0x02,0x00,0x11,0x03,0x04,0x12,0x21,0x31,0x13,0x05,0x22,0x32,0x41,0x51,0x14,0x15,0x61,0x71,0x81,0x91,0xFF,0xDA,0x00,0x0C,0x03,0x01,0x00,0x02,0x11,
0x03,0x11,0x00,0x3F,0x00,0x6C,0x4C,0xFE,0x4C,0xAB,0x2B,0x4A,0x65,0xB5,0x2B,0x01,0x74,0x58,0x90,0x24,0x30,0x13,0x6D,0xEC,0x3F,0x7B,0xD7,0x9E,0x34,0x53,0xE7,0x51,
0x8D,0x15,0x11,0x42,0xD3,0x4A,0xA7,0x50,0x91,0x9F,0x48,0x66,0xF3,0x37,0xDC,0x9F,0xA7,0xDB,0x13,0xFC,0xE3,0x3D,0xA0,0xAC,0xCC,0x84,0x39,0x74,0x46,0x34,0x43,0x76,
0x9A,0x49,0x0B,0x28,0x3E,0xE4,0x0B,0xFD,0x00,0xF7,0x38,0x6B,0xE1,0x9C,0xF3,0x86,0x29,0x20,0x66,0xAB,0xA4,0x7A,0xD9,0x2C,0x2F,0x34,0xB2,0x68,0x43,0x6F,0xF7,0x58,
0x9B,0x7A,0x2E,0x38,0xAE,0x8C,0x00,0x27,0xB9,0xE8,0x43,0x83,0xF1,0x8C,0x73,0x71,0xB5,0x5E,0x59,0x46,0x8F,0x57,0x2B,0xB9,0x67,0x2B,0xD8,0xC7,0xE1,0xD4,0x4E,0xFC,
0xB9,0x91,0xEA,0xD8,0x16,0x2B,0xF3,0xEE,0x28,0xAA,0x4E,0xF3,0x53,0xDD,0x29,0x57,0x60,0x01,0x0A,0xAA,0x3F,0x4C,0x0B,0xAA,0xCD,0x78,0x4E,0x59,0x5E,0x70,0x89,0x08,
0x0D,0x71,0x69,0x1D,0x8F,0x3E,0x40,0xB3,0x1D,0xBD,0x80,0xC6,0xCA,0x0C,0xDE,0x3A,0x97,0x58,0xF2,0xAC,0xBA,0x5A,0x89,0x01,0xBA,0xEA,0x21,0x45,0xFA,0x1F,0x15,0xB0,
0x46,0xEB,0x88,0x3C,0x5D,0xCA,0x77,0x08,0xC1,0x96,0xE5,0x71,0xDB,0x2D,0x53,0x99,0x56,0x1E,0x73,0xB8,0x2B,0x1A,0x1E,0x5F,0x31,0xDD,0xBA,0xF2,0xBF,0x2E,0x78,0x25,
0x9B,0xE7,0xD9,0x95,0x04,0xD2,0x45,0x51,0x53,0x1B,0x80,0x36,0x85,0x06,0x94,0x3E,0x44,0x8F,0xE7,0x7C,0x22,0xC3,0x98,0x71,0x7D,0x2C,0x5A,0xCF,0x78,0x90,0x91,0xA7,
0xB1,0xA0,0x84,0x39,0xFF,0x00,0xC8,0x9E,0x9E,0x98,0x5C,0xAE,0xE2,0x5A,0xE3,0x9C,0x32,0x54,0x43,0x1D,0x35,0x41,0xF9,0x9A,0xA9,0x8C,0x92,0xDB,0xD5,0x4D,0xBF,0x8E,
0x23,0x29,0x65,0xAB,0x90,0x28,0xBB,0x8F,0xA2,0xBA,0xA3,0x36,0x74,0x7A,0xFE,0xED,0x28,0x ...

The  CreateArray() that I ported to SD produces:
const uint8_t /EagleEye[] PROGMEM = {
0xFF,0xD8,0xFF,0xE0,0x00,0x10,0x4A,0x46,0x49,0x46,0x00,0x01,0x01,0x00,0x00,0xB4,0x00,0xB4,0x00,0x00,0xFF,0xE1,0x02,0x14,0x45,0x78,0x69,0x66,0x00,0x00,0x4D,0x4D,
0x00,0x2A,0x00,0x00,0x00,0x08,0x00,0x07,0x01,0x0E,0x00,0x02,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x62,0x01,0x12,0x00,0x03,0x00,0x00,0x00,0x01,0x00,0x01,0x00,0x00,
0x01,0x1A,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0x82,0x01,0x1B,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0x8A,0x01,0x28,0x00,0x03,0x00,0x00,0x00,0x01,
0x00,0x02,0x00,0x00,0x01,0x32,0x00,0x02,0x00,0x00,0x00,0x14,0x00,0x00,0x00,0x92,0x87,0x69,0x00,0x04,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0xA6,0x00,0x00,0x00,0x00,
0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x00,
0x00,0x00,0x00,0xB4,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0xB4,0x00,0x00,0x00,0x01,0x32,0x30,0x30,0x38,0x3A,0x31,0x32,0x3A,0x32,0x37,0x20,0x31,0x35,0x3A,0x31,0x36,
0x3A,0x34,0x33,0x00,0x00,0x16,0x82,0x9A,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,0x01,0xB4,0x82,0x9D,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,0x01,0xBC,0x91,0x01,
0x00,0x07,0x00,0x00,0x00,0x04,0x01,0x02,0x03,0x00,0x91,0x02,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,0x01,0xC4,0x92,0x01,0x00,0x0A,0x00,0x00,0x00,0x01,0x00,0x00,
0x01,0xCC,0x92,0x02,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,0x01,0xD4,0x92,0x04,0x00,0x0A,0x00,0x00,0x00,0x01,0x00,0x00,0x01,0xDC,0x92,0x05,0x00,0x05,0x00,0x00,
0x00,0x01,0x00,0x00,0x01,0xE4,0x92,0x0A,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,0x01,0xEC,0xA0,0x00,0x00,0x07,0x00,0x00,0x00,0x04,0x30,0x31,0x30,0x30,0xA0,0x01,
0x00,0x03,0x00,0x00,0x00,0x01,0x00,0x01,0x00,0x00,0xA0,0x02,0x00,0x04,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0xA0,0xA0,0x03,0x00,0x04,0x00,0x00,0x00,0x01,0x00,0x00,
[... cut ...]
0x00,0x01,0x00,0x02,0x00,0x00,0xA2,0x17,0x00,0x03,0x00,0x00,0x00,0x01,0x00,0x02,0x00,0x00,0xA3,0x00,0x00,0x07,0x00,0x00,0x00,0x01,0x03,0x00,0x00,0x00,0xA4,0x01,
0x00,0x03,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0x00,0xA4,0x02,0x00,0x03,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0x00,0xA4,0x04,0x00,0x05,0x00,0x00,0x00,0x01,0x00,0x00,
0x02,0x04,0xA4,0x06,0x00,0x03,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0x7D,0x00,0x00,0x00,0x39,0x00,0x00, ....

Its not the same, 

Any idea why?

sst2001, 2018-08-02:
Hello @Bodmer ,

I modified the example files and created a new code.  This is tested on TTGO VS from LylyGo and works as follows:

1. Read all files in the SD
2. Every JPG or JPEG file smaller than 250,000 bytes is converted using your function to a bit Array
3. Creates a new images.h file in the SD that contain all the bitArray and can be used to display th eimage using "drawArrayJpeg(EagleEye, sizeof(EagleEye), 0, 0);" 

Please advise how to send it to you so you can publish it.
Thx.

Bodmer, 2018-08-07:
Some jpegs seem to fail to decode and this appears to be dependent on the source program.

I tried one of your images above and it did not render, but after loading it into paint and saving it again as a jpeg it worked fine.  This appears to be an issue with the original picopeg code and I have not been able to identify the root cause.

closed 2018-08-07 by Bodmer

laoadam, 2022-10-12:
Hi,
I am testing the TFT_eSPI example of Flash_PNG, that shown panda works well, question is how to convert other png pictures into .c file to be show?
Thanks
Adam

Bodmer, 2022-10-12:
This is detailed in the header of the sketch:
https://github.com/Bodmer/TFT_eSPI/blob/master/examples/PNG%20Images/Flash_PNG/Flash_PNG.ino#L1-L14

I can provide further explanation if this is not understood.