Bodmer/JPEGDecoder #28: Progressive JPEGs not supported

schoofseggl, 2018-11-14:
Great work, I really like this lib. Its 100 times faster than the standard ST77xx lib for ESP8266.

One think should be added: Please document the limitation that progressive jpegs are not supported. It took me a while to figure out that that was my problem :)

This is my main loop. 

```
void loop()
{
  
  drawJpeg("/test1.jpg", 0, 0); //not progressive, can be displayed
  delay(5000);
  drawJpeg("/test2.jpg", 0, 0); //progressive, is not displayed
  delay(5000);

}`
```
It think there is a small bug, Obviously `JpegDec.decodeFsFile(filename); ` returns true, even though decoding failed due to the progressive jpeg.  The console produces irritating output (no error, but  0 render time, and all the meta information is  from the previous file.)

```
===========================
Drawing file: /test1.jpg
===========================
===============
JPEG image info
===============
Width      :300
Height     :300
Components :1
MCU / row  :19
MCU / col  :19
Scan type  :4
MCU width  :16
MCU height :16
===============

Total render time was    : 142 ms
=====================================
===========================
Drawing file: /test2.jpg
===========================
===============
JPEG image info
===============
Width      :300
Height     :300
Components :1
MCU / row  :19
MCU / col  :19
Scan type  :4
MCU width  :16
MCU height :16
===============

Total render time was    : 0 ms
=====================================
```

Bodmer, 2018-11-14:
Thanks for reporting this, I have updated the ReadMe.

I will clear the parameter list and consider adding an error code at a future time.

Bodmer, 2018-11-15:
Updated.

closed 2018-11-15 by Bodmer