Bodmer/JPEGDecoder #30: some white random blocks in all image using ESP32_SDcard_jpeg.ino

fotosettore, 2018-11-27:
hi
i tried to use ESP32_SDcard_jpeg.ino  but all images appear wih random blocks.
please see pictures attached
[image]
[image]
[image]
i changed sd, display, SPI_FREQUENCY (from 27 to 10) and picture definition. It does not work.
i'm using esp32 and a ili9341
i write this post here because i don't know if error is due to jpegdecoder or other
thanks for your help

Bodmer, 2018-11-27:
Are you using the TFT_eSPI library or another graphics library?

I would guess you are using TFT_eSPI in that case [run this example](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/160%20x%20128/TFT_flash_jpg), which reads the Jpeg image from arrays stored in program memory. And [this example ](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/480%20x%20320/TFT_flash_jpg)which is for a bigger display.

You will need to set the CS line for the SD Card to high, or remove the SD Card to run these examples to avoid the SD card driving the SPI bus.

Run the graphics test to see if the display interface is working correctly.

Do the white blocks appear over a correct image or are those blocks always white if the image is repainted without clearing the screen?

Are the white blocks always in the same place on the same image?

Bodmer, 2018-11-28:
Both the examples linked to above use the jpeg decoder, you probably ran the wrong sketch. Make sure the sketch name you ran ends in _jpg

The request to run these is not to show an alternative storage method but to show if the FLASH jpeg examples work and decode and display correctly.  This demonstrates tehrefore that the JPEGDecoder library must be working OK with your hardware and your interface to the TFT is reliable. This means the problem must lie with the sketch or the SD Card interface, or the SD library.  In this case there is little I can do to advise you as I have run the sketch and it works fine on my setup with and ESP32.

The blocks you see as whith are 16x16 pixel "Minimum Coding Units", it is likely that in each one of these blocks there is a bad byte which breaks the decoding. It could be that power cycling your setup or inserting and removing the card with the setup powered has corrupted each file.

Possible causes:
1. look at the power supply and see if it is adequate to power your hardware
2. change the SD card to one of a different make, some cards are less reliable
3. put the SD card in a PC and see if the files have been corrupted, fully format the SD card (not quick format) and renew the image files
4. check if there are any reported issues with the SD library and the ESP32
5. inserting and removing an SD card when the setup is powered and working sometimes corrupts the card files, file access must be stopped and the SPI bus must not be transferring data when you insert and remove the card other wise the SD Card may see garbage erase commands

fotosettore, 2018-11-28:
i changed all hardware : esp32, card reader, card (from 2gb to 8 gb, formatted slow in fat32), display, images.  i tried other images, smaller and bigger .... 
all new but no lucky : the same problem, and in the oldf images the same blocks in the same places :-(

so, as you said, i think problems is elsewhere. it raises to see if the reason for the failure is due to the sofware

may i post here my little sketch so you can try it or there is a different way ?(i'm new on github, I would not want to go wrong with the rules).

Bodmer, 2018-11-28:
You can post zip files to github with the message, so that is the easiest way. I will need the images as well (in the same zip file would make this easier), so that I can be sure the decoder is doing the same maths. Also which gpio pins you are using, so a copy of the setup file, which Arduino IDE version and which ESP32 board support package version you are using.

Bodmer, 2018-11-28:
I think I know what the problem is...

I cannot see the TFT connections clearly but it looks like your TFT does not use a CS (chip select) signal, this means it cannot share the same SPI bus as the SD card because it will see the data transfers between the SD card and the ESP32 which, depending on the data pattern, will corrupt the current display block. I am surprised it works at all.

All devices that share an SPI bus must have their own chip select.

fotosettore, 2018-11-28:
no .. it does not work :-(
is there a way to overcome the problem of the missing CS?
I'm curious to understand why it does not work and in the meantime I have ordered this card that has the CS : 
https://www.amazon.it/gp/product/B0749NHRV4/ref=ox_sc_act_title_1?smid=A2F7JN9V2AWTSH&psc=1.
it will be delivered next friday

fotosettore, 2018-11-28:
I found the same display in a shop in my town. 
I connected and ... all works fine !
My compliment to you, dear Bodmer :+1:  
You discovered that ili9341 without CS cannot work with a SD reader.
I don't know very well the "inside world" of arduino so without your help no way out for me.

I still have doubts about how it is possible to sell a display without such an important connection.
and, again, is there a workaround to be able to simulate CS?

Bodmer, 2018-11-28:
It would be possible to add some circuitry between the display and ESP32 but it would need some knowledge of electronics. Essentially you would need to prevent the SCK getting to the display while accessing the SD card, this could be done with an AND gate or equivalent circuit to keep the clock line to the display low while the SD card is being accessed. If you can tolerate a longer wait for displays from China (3-4weeks) then they are available from Aliexpress, so that is where I get mine.

The display without the chip select will still work but cannot share the SPI bus with anything else so you may still find it is useful on another project, [like this Weather station]( There are work-arounds but that then needs detailed software knowledge and modification of the library.) library, which just needs a display and a WiFi internet connection to work.

Glad to hear you have it working.

closed 2018-11-28 by Bodmer