Bodmer/JPEGDecoder #31: Picture mess on 80*160

Celppu, 2018-11-28:
I have smaller ips 80*160 rgb display and i cannot get decoder to work correctly. I can see that it draws 16x16 blocks but those are in somehow rnd order and messy. I'm using nodemcu_jpeg example

Celppu, 2018-11-28:
Öööh it seems that almost same block is multiple times on screen. What causes this?

Bodmer, 2018-11-29:
The example is for a 240 x 320 screen.

You must specify the width (i.e. 80) and height (i.e. 160) of your screen in the setup file so the jpeg decoder function knows where the edge of the screen is.

Celppu, 2018-11-29:
But in the jpeg funtions tft.width() and tft.height() returns resolution already? Or setup file somwhere else?

Celppu, 2018-11-29:
It seems that it won't stay in boundries set with setAddrWindow. I wrote version that uses drawPixel and it is 3x slower....

Bodmer, 2018-11-30:
I am guessing your are using the TFT_eSPI library.

The option to use in the setup file is:

#define ST7735_REDTAB160x80
or
#define ST7735_GREENTAB160x80

I have just tested my display and though the screen protector has a green tab I had to use the 2nd option above.  When you use this option the screen size is set for you so you do not need to specify the width and height in the setup file.

Bodmer, 2018-11-30:
Actually you do need to define the width and height in setup...

#define TFT_WIDTH  80
#define TFT_HEIGHT 160

closed 2019-01-21 by Bodmer