Bodmer/JPEGDecoder #33: Not able to display images on ILI9341 240*320 Tft by using ESP8266

acharyanachiket, 2019-01-06:


Bodmer, 2019-01-06:
If the images look rather psychedelic then it is likely the byte order swap needs to be set in your sketch.

Please identify which example you are trying to run and which graphics library you are using.

closed 2019-01-21 by Bodmer

acharyanachiket, 2019-01-28:
uncommented SWAP_BYTES in user_config.h and also used readSwappedBytes() instead of read() still see the images totally blurred(parallel lines). Tried changing the SPI frequency too. 
Sketch used is NodeMCU_Jpeg and TFT I use ILI9341
[image]

acharyanachiket, 2019-02-06:
> If the images look rather psychedelic then it is likely the byte order swap needs to be set in your sketch.
> 
> Please identify which example you are trying to run and which graphics library you are using.

uncommented SWAP_BYTES in user_config.h and also used readSwappedBytes() instead of read() still see the images totally blurred(parallel lines). Tried changing the SPI frequency too.
Sketch used is NodeMCU_Jpeg and TFT I use ILI9341. I amtrying to display BaboonP image here
[image]

Bodmer, 2019-02-09:
This turned out to be a change made by the Adafruit setAddrWindow() function parameters.

I have updated the library examples for Adafruit_GFX so you should find it is working now.

Thanks for reporting this.