Bodmer/JPEGDecoder #37: Adafruit_GFX compatibility lost

Bodmer, 2019-01-21:
The Adafruit_GFX library has been updated and may no longer be compatible with this library.

If the example sketches for the Adafruit library no longer work for you then post details here please.

acharyanachiket, 2019-02-06:
> The issue is due to a breaking change in setAddrWindow. It used to take x1, y1, x2, y2 but now it takes x1, y2, width, height.
> 
> Also it's necessary to preface your write with a startWrite() and end it with an endWrite().
> 
> Example:
> 
> Before:
> 
> ```
>     // draw image MCU block only if it will fit on the screen
>     if ( ( mcu_x + win_w) <= tft.width() && ( mcu_y + win_h) <= tft.height())
> 	{
>       // Now set a MCU bounding window on the TFT to push pixels into (x, y, x + width - 1, y + height - 1)
>       tft.setAddrWindow(mcu_x, mcu_y, mcu_x + win_w - 1, mcu_y + win_h - 1);
> 
>       // Write all MCU pixels to the TFT window
>       while (mcu_pixels--) tft.pushColor(*pImg++); // Send MCU buffer to TFT 16 bits at a time
>     }
> ```
> After:
> 
> ```
>     tft.startWrite();
> 
>     // draw image MCU block only if it will fit on the screen
>     if ( ( mcu_x + win_w) <= tft.width() && ( mcu_y + win_h) <= tft.height())
> 	{
>       // Now set a MCU bounding window on the TFT to push pixels into (x, y, x + width - 1, y + height - 1)
>       tft.setAddrWindow(mcu_x, mcu_y, win_w, win_h);
> 
>       // Write all MCU pixels to the TFT window
>       while (mcu_pixels--) tft.pushColor(*pImg++); // Send MCU buffer to TFT 16 bits at a time
>     }
> 
>     tft.endWrite();
> ```
> I tested this and it seems to work on my 1.5" ST7789 Adafruit TFT.

Hi John
This fix of yours works fine on nodeMCU_Jpeg.ino too, wherein we interface it with ILI9341 driver. Thank you for the update

john--, 2019-02-06:
@acharyanachiket good to hear! I would submit a PR to update the examples, but I just don't have the time to test each of them.

geeksville, 2019-02-08:
@acharyanachiket thank you for your note it helped me a lot. (John, if you want a PR I can send one in)

john--, 2019-02-08:
@geeksville it's not my repo but I think the community would find this helpful :) sounds like we've confirmed with enough people that the suggestion seems to resolve the issue.

Bodmer, 2019-02-09:
Thanks for your input. I have updated the Adafruit_GFX examples.

The examples also handle the case where the width and height of the image is not an integral number of mcu blocks.

I have implemented the change using drawRGBBitmap(), this may make rendering a bit faster.

closed 2019-02-09 by Bodmer

Bodmer, 2021-05-10:
The example is for the Arduino TFT library (hence the  "#include <TFT.h>") which contains an older (2017) version of Adafruit_GFX, this old version does not provide a drawRGBBitmap() function so the above suggestion will not work without changing the graphics library.

I assume therefore you have made other changes to the sketch to use the Adafruit_GFX and Adafruit_ILI9341 libraries? Please detail these changes.