Bodmer/JPEGDecoder #40: Error 18 (PJPG_UNEXPECTED_MARKER) in jpeg_kbv

Masi2017, 2019-03-20:
Hi @Bodmer 
I am trying to use this library in my code ( not in Arduino environment) .
For now I am trying to test  jpeg_kbv and convert the existing Baboon jpeg image data in jpeg1.h. I'm getting error number 18 ( status = 18 in  locateSOSMarker). I'd appreciate if you can help me with that.

Bodmer, 2019-03-20:
This error originates in the picojpeg code which I did not write and the error cause is a mystery to me.

What hardware are you running it on?

Masi2017, 2019-03-21:
@Bodmer I am using K210 from Kendryte , For now I am only testing the converter using the sample jpeg in jpeg1.h. So What can I do about this? This means that , it may not be able to convert some of the images?

Bodmer, 2019-03-21:
The decoding works for other processors so I do not know why you get this error or how to fix this.

closed 2019-03-21 by Bodmer

Masi2017, 2019-03-21:
@Bodmer Would you please just let me know if you have tested jpeg_kbv project? I want to make sure if I have done something wrong in porting or this has error in converting this image so I can't do anything about it.

Bodmer, 2019-03-22:
The example runs fine without any errors on an Arduino Due.

Masi2017, 2019-03-22:
OK thanks, My controller is 64bits and this may causes the problem. I'll figure that out.