Bodmer/JPEGDecoder #49: multi-thread decoding

Dimision, 2020-04-23:
As well known that ESP32 has dual cores, and Arduino code usually runs on core 1, is there any possibility to use both core 0 and 1 to decoding jpg to improve performance?

Bodmer, 2020-04-23:
Yes,

You can improve performance by using the [TJpg_Decoder library](https://github.com/Bodmer/TJpg_Decoder) on one processor core. The TJpg_Decoder library uses 32 bit variables and is more suited to the ESP32. The library has an example Flash_Jpg that will run on an ESP32.

You can use both processors, the attached sketch is a copy of the Flash_Jpeg sketch which is modified to use both processors. The sketch takes 103ms on a single processor core, this total is comprises 66ms to decode the Jpeg and 37ms to render to screen on an ILI9341 at 40MHz SPI rate. When the second core is used for decoding and the standard sketch core is used for drawing on the TFT the Jpeg time to decode and render is 66ms. Thus decoding and rendering are being done in parallel.

To use both processors a knowledge of mutex flags and the importance of correct flag management must be understood.

[ESP32_Dual_Core_Flash_Jpg.zip](https://github.com/Bodmer/JPEGDecoder/files/4525863/ESP32_Dual_Core_Flash_Jpg.zip)

closed 2020-04-30 by Dimision

Celppu, 2022-03-09:
Hi would it be possible to decode image in parts? I have a use case where i could use both cores or multiple esp32 to decode image as fast as possible.

Bodmer, 2022-03-10:
You would have to split the jpeg into smaller sections in separate image files, and then join them to create the final image. A single jpeg has to be decoded serially from the start. You could write your own decoder that isolates the MCU block boundaries in one task and decodes in another  but you would need a deep understanding of the jpeg compression format.