Bodmer/JPEGDecoder #55: Incoherent  "y + JpegDec.MCUHeight*mcuY"

Barrois, 2020-11-07:
Hi,
I would like to detect an Asian hornet in a photo taken by Esp32-Cam, 
but when I decode Jpegs there are sometimes incomprehensible jumps in the "y + JpegDec.MCUHeight * mcuY":
    Serial.print (s_s_x);
    Serial.print (",");
   Serial.print (s_s_y);
  // Serial.print ("and its color is:");
    Serial.print (",");
    Serial.println (s_gris);
gives:
13, 36, 248
14, 36, 248
15, 36, 248
0, 37, 248
1, 3973, 248
2, 3973, 248
3, 3973, 248
is this normal?
Why these sudden jumps?
is this an effect of jpeg encoding ?
Do we have to count the coordinates separately?
Thank you for your reply !

Bodmer, 2020-11-07:
I suspect there is a bug somewhere in your sketch. Without seeing more of your code I can only guess where. One possibility is that you are running out of memory and variables are being corrupted.

Barrois, 2020-11-07:
> Message du 07/11/20 13:54
> A : "Bodmer/JPEGDecoder"
> Copie à : "Barrois" , "Author"

I hope that

Thanks !

Bernard BARROIS

15 rue de la revenue
70400 Chenebier
tel : 03 84 27 30 19
tel : 06 28 13 62 47

closed 2020-11-07 by Bodmer

Barrois, 2020-11-08:
> Message du 08/11/20 00:32

Oh, I'm a little disappointed with your short answer, 
could you take a look at my code for a few moments 
and give me some leads? 
Thanks in advance

Bernard BARROIS

15 rue de la revenue
70400 Chenebier
tel : 03 84 27 30 19
tel : 06 28 13 62 47

Bodmer, 2020-11-08:
I suspect there is a bug somewhere in your sketch. Without seeing more of your code I can only guess where. Post a complete sketch.