Bodmer/JPEGDecoder #56: Incohérent "y + JpegDec.MCUHeight * mcuY" with code

Barrois, 2020-11-08:
[image]
[image]

Here is my code

// https://www.deviceplus.com/arduino/jpeg-decoding-on-arduino-tutorial/
// Include the library
#include <JPEGDecoder.h>
#include <FS.h>               // SPIFFS
// this function determines the minimum of two numbers
#define minimum(a,b)     (((a) < (b)) ? (a) : (b))

// The coordinate variables
int x, y, mcuX, mcuY;
// The color of the current pixel
int inColor, r, g, b;
unsigned int s_s_x, s_s_y;      // sauve les coordonnées du point noir
unsigned int tableau[32]; // deux fois 4 x 4
unsigned int quelle_carre , carre_1, carre_2;
unsigned int x_noir, y_noir, noir ;
void setup() {

  // Begin Serial port for communication with PC
  Serial.begin(115200);
  Serial.println(__FILE__);  
   Serial.println(".........................init");
    
    //Initialize SPIFFS
  Serial.println("Initializing SPIFFS... ");
  SPIFFS.begin();
    if(!SPIFFS.begin()){
    Serial.println("SPIFFS  Mount Failed");
    return;
  }
   char NOM_jpg[] = "/Image7.jpg";
    Serial.printf("Picture file name: %s\n", NOM_jpg);
    
    File jpgFile = SPIFFS.open(NOM_jpg,"r");
    if(!jpgFile){
    // File not found 
    Serial.println("Failed SPIFFS to open file");
    return;
   }
  Serial.println("...................open....JPGFILE");
  if(!jpgFile){
    Serial.println("Failed to open jpgFile");
  } 
    // Decode the JPEG file
    JpegDec.decodeSdFile(jpgFile); //................ l'image est decodée
    
    Serial.print(".............image decodee : ");
    Serial.print( JpegDec.width) ;
    Serial.print(" x ") ;
    Serial.print("Height    :");
    Serial.println(JpegDec.height);
    Serial.print("Components:");
    Serial.println(JpegDec.comps);
    Serial.print("MCU / row :");
    Serial.println(JpegDec.MCUSPerRow);
    Serial.print("MCU / col :");
    Serial.println(JpegDec.MCUSPerCol);
    Serial.print("Scan type :");
    Serial.println(JpegDec.scanType);
    Serial.print("MCU width :");
    Serial.println(JpegDec.MCUWidth);
    Serial.print("MCU height:");
    Serial.println(JpegDec.MCUHeight);
    Serial.println("");    
    
     // Create a buffer for the packet
    char dataBuff[240];

    // Fill the buffer with zeros
    initBuff(dataBuff);

    // Pointer to the current pixel
    uint16_t *pImg;      
    // Color of the current pixel
    uint16_t color;    
      
        // Repeat for all MCUs in the image...............
    while(JpegDec.read()) {
          Serial.print(".............MCU : ");
          Serial.println(JpegDec.read());          
      // Save pointer the current pixel
      pImg = JpegDec.pImage;
      
      // Get the number of pixels in the current MCU
      uint32_t mcuPixels = JpegDec.MCUWidth * JpegDec.MCUHeight;
          Serial.print(".................mcuPixels : ");
          Serial.println(mcuPixels);    
      // Repeat for all pixels in the current MCU
      while(mcuPixels--) {
        // Read the color of the pixel as 16-bit integer
        color = *pImg++;
     
      // Convert 16-bit color into RGB values
      r = ((color & 0xF800) >> 11) * 8;
      g = ((color & 0x07E0) >> 5) * 4;
      b = ((color & 0x001F) >> 0) * 8;
      
// https://onlinejpgtools.com/convert-jpg-to-grayscale
// 0.21 r + 0.72 b + 0.07 v Méthode BT-709
      
      int s_gris = 0.21 * r + 0.72* b +0.07 * g; // nuance de gris
        s_s_x = x + JpegDec.MCUWidth*mcuX ;      // on garde les coordonnées
        s_s_y = y + JpegDec.MCUHeight*mcuY;      //  s_s_x et s_s_y 

tableau[s_s_x,s_s_y] = s_gris ;
   
 //   Serial.print("+++++++++++++ point le plus noir est en : ");
    Serial.print(s_s_x);
    Serial.print(" , ");
//    if (s_s_y > 2640 ) s_s_y = s_s_y - 2640; // ????? du à la copie
    Serial.print(s_s_y);
  //  Serial.print(" et sa couleur est : "); 
    Serial.print(" , ");     
    Serial.println (s_gris);   
  //  }
  //  }

      // Move onto the next pixel
      x++;
      
      if(x == JpegDec.MCUWidth) {
        // MCU row is complete, move onto the next one
        x = 0;
        y++;
      }
      
      if(y == JpegDec.MCUHeight) {
        // MCU is complete, move onto the next one
        x = 0;
        y = 0;
        mcuX++;
      }
      
      if(mcuX == JpegDec.MCUSPerRow) {
        // Line of MCUs is complete, move onto the next one
        x = 0;
        y = 0;
        mcuX = 0;
        mcuY++;
      }
      
      if(mcuY == JpegDec.MCUSPerCol) {
        // The entire image is complete
        return;
      }
    }
  }
}

// Function to fill the packet buffer with zeros
void initBuff(char* buff) {
  for(int i = 0; i < 240; i++) {
    buff[i] = 0;
  }
}

void loop() {
  // Nothing here
  // We don't need to send the same images over and over again
}
int getMin(int* array, int size){
int minIndex = 0;
 int min = array[minIndex];
 for (int i=1; i<size; i++){
//   if (min<array[i]){             // What you wrote
   if (array[i] < min){             // What Rob wrote
     minIndex = i;
     min = array[i];
   }
 }
 return minIndex;
}

Bodmer, 2020-11-08:
It looks like you do not initialise the variables correctly so they are undefinied initially... use:
```
int x = 0, y = 0, mcuX = 0, mcuY = 0;
```

Barrois, 2020-11-09:
Yes !
You are right !
in fact it is in the definition of the table, which has nothing to do ...
Thanks a lot for your help.

closed 2020-11-09 by Barrois