Bodmer/JPEGDecoder #60: Does not work on ESP32 when choosing Littlefs

aleXela17, 2021-04-13:


aroesz98, 2021-07-24:
Is it possible to use this library with Arduino Nano RP2040 Connect and arduino pico library? I'm trying to use this library with LittleFS but display only change colors and bitmaps doesn't printing.

aroesz98, 2021-07-24:
Actually after changing all "SPIFFS" lines to "LittleFS" showed me something like that for all images:
Drawing file: /BaboonL.jpg
=====================================
===============
JPEG image info
===============
Width      :536886536
Height     :536886776
Components :268445121
MCU / row  :536886800
MCU / col  :536887976
Scan type  :193
MCU width  :268550808
MCU height :1073987584
===============
Total render time was    : 1 ms
=====================================