Bodmer/JPEGDecoder #78: Example code for UTFT, UTFT_SD.Jpeg does not work for LANDSCAPE

drp0, 2023-04-29:
The example works perfectly for all of the supplied images in PORTRAIT.
If myGLCD.InitLCD(LANDSCAPE); is used the image drawn is not correct.
Is this a problem with renderJPEG ?
Can a landscape alternative be provided?

David

drp0, 2023-05-02:
I am using a 320x240 screen, with utft:

```
UTFT myGLCD(ITDB32S_V2,38,39,40,41);  // latest Henning UTFT needed
// myGLCD(model, SDA, SCL, CS, RST [, RS] )
#define TFT_CS    40
#define SD_CS     53
```

The issue appears to revolve around renderJPEG and the alignment of the bit array pImg.
I can reduce the lanscape error by rotating each pImg array clockwise by 90 degrees before pushing onto the lcd:

Test Image (320x240 px):
[image]

Portrait mode left:
[image]

Portrait mode right:
[image]

Landscape no rotation:
[image]

Landscape with rotated bitmaps:
[image]

Some imrovement shown so possibly on the right path.

My code:
Includes image centering, for x=-1 or y =-1.

```
void jpegDraw(const char* filename, int x, int y) {
byte wstate;

  // Try to open requested file on SD card
File jpegFile = SD.open(filename, O_READ); 
  if (!jpegFile) {
  Serial.println("Jpeg file not found on SD card.");
  return;
  }
jpegFile.close();

Serial.print(F("Decoding  image '"));
Serial.print(filename);
Serial.println('\'');

// initialise the decoder, check compatibility and gain access to image information
boolean decoded = JpegDec.decodeSdFile(filename);

  if (decoded) {
  // print information about the image to the serial port
  jpegInfo();
  
  int X  = JpegDec.width;
  int Y  = JpegDec.height;
  int XX = myGLCD.getDisplayXSize();
  int YY = myGLCD.getDisplayYSize();
  bool sx         = X < XX;
  bool bx         = X > XX;
  bool sy         = Y < YY;
  bool by         = Y > YY;

    if (filename != "arduino.jpg") {
      // if (JpegDec.width > JpegDec.height) wstate = LANDSCAPE; else wstate = PORTRAIT;
      
      // if (page != wstate) {
      // myGLCD.InitLCD(wstate);
      // page = wstate;
[... 164 lines cut ...]
  if (left){                                                 // 2d to 1d
  p = 15;
    for (r = 0; r < size; r++) {
      for (c = 0; c < size; c++) {
      in[p] = m[r][c];
      p--;
      }
    }
  }else{
  p = 0;
    for (r = 0; r < size; r++) {
      for (c = 0; c < size; c++) {
      in[p] = m[r][c];
      p++;
      }
    }
  }
}

```
Rotating the bit image anticlockwise does not improve the result.  
I am out of ideas, now.

Any suggestions?
David

Bodmer, 2023-05-02:
The problem is that UTFT does not use the hardware rotation features built into the display. This is possibly because it supports some displays that do not have this feature. This ultimately means that the display is kept in portrait orientation but x and y for each pixel are swappped (if required) and coordinates adjusted (if required) by subtracting from the width and height of display.

There are a few ways to deal with this:
1. Extract each pixel from the decoded MCU block image (typically 8x8 or 16x16) then plot the pixels on the screen in the correct place.
2. Rotate the image block and swap/adjust mcu_x, mcu_y (if required) n the myGLCD.setXY() function.
3. Over-ride the UTFT library by setting the width and height in the desired rotation then force the hardware rotation feature of the display to the same rotation. In this case don't tell UTFT library to perform the rotation.
4. Command UTFT to rotation 0, then display an image that has already been rotated (e.g. 0, 90, 18 or 270 to look correct), then swap back to the desired rotation for future graphics operations.

Personally I would avoid the UTFT library and use TFT_eSPI or Adafuit_GFX. Those libraries make thing simpler and use hardware rotation and have a performance advantage. Performing the rotations in software for every pixel has a performance impact, but does make the library "universal" which is as the author intended.

drp0, 2023-05-03:
Thanks for the reply.
I have around 6 tft units, none of which appear to be compatible with the TFT_espi or adafruit GFX libraries.

Does landscape work using the library example with any hardware combinations that you have tested?

I am using a mega with a ITDB32S_V2 screen.

Re 1 & 2, As detailed above I have extracted the MCU block and rotated it, with some improvement with simple lined shapes.
With detailed pictures, the screen image is barely recognisable. 

Re 3 & 4, I had already tried the pre-rotating trick but this would not be a satisfactory solution.

Is JpegDec.MCUx properly tied into the screen alignment, or is it specifically returning portrait positions?
At what point is JpegDec.MCUx updated?
Does the JpegDec.read() function read in a linear book-reading fashion through the jpeg (in x by y blocks) ?

David

Bodmer, 2023-05-03:
Intuitively, looking at the image where blocks are not rotated, a 90 degree anticlockwise rotation of each MCU block will correct the image. What does it look like?

drp0, 2023-05-03:
90 degree anticlockwise was a good call.
I also had to rewrite the anticlockwise rotation section using google code.
I have centreing and positioning working, with options to auto-rotate the screen for portrait / landscape pictures.
Here is the full code. You are welcome to use it as you see fit.

David

main:

```
/*
UTFT_SD_Jpeg4.ino
Modified jpeg decoder
D.R.Patterson
3/5/2023

 This sketch draws Jpeg files stored on an SD card on a TFT screen, it is based on the UTFT_Bitmap
 by Henning Karlsen. It is currently limited to PORTRAIT.

 web: http://www.RinkyDinkElectronics.com/

 The demo should also run on an Arduino Mega, but it will be slower

  By default the UTFT library does not configure the Gamma curve settings for the ILI9341 TFT,
  so photo images may not render well.  To correct this ensure the set Gamma curve section
  in initlcd.h (library folder UTFT\tft_drivers\ili9341\s5p\initlcd.h) is NOT commented out.

  You can generate your own Jpeg images from digital photographs by cropping and resizing
  by using commonly available picture/image editors such as Paint or IrfanView.

  The latest JPEGDecoder library can be found here:
  https://github.com/Bodmer/JPEGDecoder

  Information on JPEG compression can be found here:
  https://en.wikipedia.org/wiki/JPEG 
  
====================================================================================
  libraries
====================================================================================
*/

#include <SPI.h>
#include <SD.h>  // Use the Arduino IDE built-in SD library

#include <UTFT.h>

UTFT myGLCD(ITDB32S_V2,38,39,40,41);                        // For new screens latest Henning UTFT needed
// myGLCD(model, SDA, SCL, CS, RST [, RS] )
#define TFT_CS    40                                        // Chip Select for TFT
extern uint8_t SmallFont[];                                 // Declare which fonts we will be using 

#include <JPEGDecoder.h>                                    // JPEG decoder library

// SD card connects to hardware SPI pins MOSI, MISO and SCK and the following chip select
#define SD_CS 53                                            // Chip Select for SD card
#define defaultselect 53                                    // Mega default select

char * files[] = {"cross.jpg", "david.jpg", "tiger.jpg", "/xsail/mast2.jpg", "Baboon20.jpg", "Baboon40.jpg", "EagleEye.jpg", "lena20k.jpg", "Mouse480.jpg" };
                    
unsigned int Ly;
unsigned int N = sizeof(files)/sizeof(*files);              // Number of filenames

// ===================================================================================
// Options
// ===================================================================================

byte page             = LANDSCAPE;                           // PORTRAIT / LANDSCAPE
const bool autoOrient = true;                                // if true align screen to picture orientation
const bool doOver     = false;                               // add arduino logo
int posx              = -1;                                  // x position, -1 for centred
[... 324 lines cut ...]
    tmp = m[i][j];
    m[i][j] = m[j][i];
    m[j][i] = tmp;
    }
  }

p = 0;                                                      // 2d to 1d for left rotation
  for (j = 0; j < size; j++) {
    for (i = 0; i < size; i++) {
    in[p] = m[j][i];
    p++;
    }
  }
}

void mtest(){
// matrix rotation test
int q[16] = {1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16};
Rmatrix(q, 16);
  for (int p = 0; p < 16; p++) {
  Serial.print("\t"); Serial.print(q[p]);
  }
Serial.println();
}
```