Bodmer/JPEGDecoder #79: Library example can not be adapted for SdFat.h

drp0, 2023-05-07:
I have tried converting the library utft example to SdFat.

I edited User_Config in JPEGDecoder\src to allow #define LOAD_SDFAT_LIBRARY.
I modified the example code to work with the latest SdFat version.
The code will not compile with all errors pointing to SD.open in JPEGDecoder.cpp:
```
c:\Users\User\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'int JPEGDecoder::decodeSdFile(const char*)':
c:\Users\User\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:353:17: error: 'SD' was not declared in this scope
  File pInFile = SD.open( pFilename, FILE_READ);
                 ^~
c:\Users\User\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:353:17: note: suggested alternative: 'SDA'
  File pInFile = SD.open( pFilename, FILE_READ);
                 ^~
                 SDA
c:\Users\User\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'int JPEGDecoder::decodeSdFile(const String&)':
c:\Users\User\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:362:17: error: 'SD' was not declared in this scope
  File pInFile = SD.open( pFilename.c_str(), FILE_READ);
                 ^~
c:\Users\User\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:362:17: note: suggested alternative: 'SDA'
  File pInFile = SD.open( pFilename.c_str(), FILE_READ);
                 ^~
                 SDA
```

Basically SdFat does not recognise SD.open.. Should it be:
```
File32 someHandle;
bool fok = somehandle.open(someFilename, O_RDONLY);
```
David

drp0, 2023-05-17:
I now have a <Sdfat.h> version working
JPEGdecoder.cpp has a couple of references to parameters passed as File type. This limits what you can do without serious rewriting. Certainly the use of fat32 would require a re-write.
It is possible to use Sdfat with..

#define SPI_CLOCK SD_SCK_MHZ(50)
#define SD_CONFIG SdSpiConfig(sdSelect, DEDICATED_SPI, SPI_CLOCK)
SdFat sd;                                     // not SdFat32 sd

  if (!sd.begin(SD_CONFIG)) {
  Serial.println(F("sd.begin failed"));
  return;
  }else  Serial.println(F("SD ok"));

File something;                              // not File32 something

Note that JPEGDecoder.cpp does not differentiate between fat and sd. I suggest the following amendment which allows decodeSdFile to work:
```
#if defined (LOAD_SD_LIBRARY) || defined (LOAD_SDFAT_LIBRARY)

// Call specific to SD filing system in case leading / is used
int JPEGDecoder::decodeSdFile(const char *pFilename) {
	File pInFile;
		#if defined (LOAD_SD_LIBRARY) 
		pInFile = SD.open( pFilename, FILE_READ);
		#endif
		//drp
		#if defined (LOAD_SDFAT_LIBRARY)
 		pInFile.open(pFilename, O_RDONLY);
		#endif 
	return decodeSdFile(pInFile);
	}

int JPEGDecoder::decodeSdFile(const String& pFilename) {
#if !defined (ARDUINO_ARCH_SAM)
	File pInFile;
		#if defined (LOAD_SD_LIBRARY) 
		pInFile = SD.open( pFilename.c_str(), FILE_READ);
		#endif
		// drp
		#if defined (LOAD_SDFAT_LIBRARY)
		pInFile.open(pFilename.c_str(), O_RDONLY);
		#endif
	return decodeSdFile(pInFile);
#else
	return -1;
#endif
}
```

David

Bodmer, 2023-05-17:
Thanks for the example code.

closed 2023-05-17 by Bodmer