Bodmer/JPEGDecoder #81: Differend colors with differend jpeg compressonrate. How to use right?

ctc-nick, 2023-07-08:
Hi! This Library works right out of the box! Thanks @Bodmer!   (ESP32 WROOM with ILI9341 preinstalled, 320*240) 
I'm using SDJpeg example, and I also want to use SD in my project.

However its hard to control which Jpeg loads fine. Some wont load at all, some partly, some change the color-grade and some just work perfect. I am trying to make 'stable' jpegs for the lib. How can I archive that?

With 24bit you probably mean '8Bit' JPEG in Image Editors 8*8*8 RGB? I tried 'Paint.net' and 'Krita' Image editors, using less colors and compression rates between 20% and 5%.  Best worked filesizes between 6kb and 17kb but it is very random, and not so easy with all the sd card copying. I also tried the TJepeg, but I couldnt make it work right now. Maybe PNG is better? 
All the Best! 👍

Bodmer, 2023-07-08:
This may be due to the screen and SD card being on the same SPI bus?  Another user told me a newer version of the SD library fixed a bug in the transaction handling in the ESP32 SD library but I have not checked that.

The library retrieves image data from the file in chunks, displays any decoded blocks and repeatscto the end of the file. It sounds like the file stream is getting corrupted so jpeg decoding goes awry.

If the SD card is on the same SPI bus, do you pass the TFT SPI instance to SD.begin? This avoids clashes in the bus state.

Connecting the display and SD card to seperate SPI busses should work.

Bodmer, 2023-07-08:
Example where TFT SPI instance is passed to SD library with chip select for SD on GPIO 22:.
```
tft.init()
if(!SD.begin(22, tft.getSPIinstance())){
    Serial.println("Card Mount Failed");
    return;
}
```

ctc-nick, 2023-07-10:
Hi! Thank you for the quick reply.

I have this Board. ESP32_2432S028_R, the SD_CS is 5, TFT_CS is 15 and Touch_CS Is 33. 
Schematics are here: https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/OriginalDocumentation/5-Schematic/ESP32-2432S028-LCM.jpg, 
and for the SD here: https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/OriginalDocumentation/5-Schematic/ESP32-2432S028-MCU.jpg (thanks @wittnesmenow!)

I think i set it properly. 
```cpp
#define ESP32
#define SD_SCK GPIO_NUM_18
#define SD_MISO GPIO_NUM_19
#define SD_MOSI GPIO_NUM_23
#define SD_CS GPIO_NUM_5
//#############################
// Setup
//############################
void setup() {
  Serial.begin(115200);

  //Set all chip selects high to avoid bus contention during initialisation of each peripheral
  digitalWrite(33, HIGH); // Touch controller chip select (if used)
  digitalWrite(15, HIGH); // TFT screen chip select
  digitalWrite( 5, HIGH); // SD card chips select, must use GPIO 5 (ESP32 SS)
  ```
[image]
This is the jpg on SD.
[image]
And this is the output on the tft.

I tried adding `(!SD.begin(SD_CS, tft.getSPIinstance()))`, but then it doesnt load any picture, just the random color. I should mention that drawing rectangles and text etc is working without problems.
I experimented by adding values to mcu_w and others, but I must admit I really dont know how i should trace that error.
Thanks again!

ctc-nick, 2023-07-12:
Hello! Update:
I reinstalled the Jpegdecoder Library and edited my userconfig.h, like shown in this threat: https://github.com/Bodmer/TFT_eSPI/discussions/2717. Then the images are loaded correct, and I also got Tjpeg Lib to work. But then I lost touch input..😜  I think I got touch working again by enabling `#define SUPPORT_TRANSACTIONS`, now it works without it. (?)  

@Bodmer : I think you are right that I have an SPI Problem. I have to learn how to share VSPI/HSPI for Touch,TFT and SD. 
It is working now, so I mark this as solved. Thanks again

my userconfig.h from 'copyCATandpasteDOG'(thanks) , maybe it helps others:
```cpp
// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_INFO "User_Setup"

// Define to disable all #warnings in library (can be put in User_Setup_Select.h)
//#define DISABLE_ALL_LIBRARY_WARNINGS

// ##################################################################################

#define TFT_WIDTH  240 
#define TFT_HEIGHT 320 

#define TFT_BL   21            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

// ######      TFT & SD CARD & ESP32 (2 SPI MODULES)      ######

#define ILI9341_DRIVER

//TFT Pins (HSPI)
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (or could connect to RST pin if you want)
//#define TFT_RST  -1  // Set TFT_RST to -1 ( if display RESET is connected to ESP32 board RST)

//Touch Pins (HSPI)
#define TOUCH_DO 12 
#define TOUCH_DIN 13 
#define TOUCH_CLK 14
#define TOUCH_CS 33//27 // Chip select pin (T_CS) of touch screen//choose what you want

//SD Card Pins (VSPI)
#define SD_MISO 19 //SD PINS
#define SD_MOSI 23
#define SD_SCLK 18
#define SD_CS   5   //i prefered to use this pin             

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  20000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000 // The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define USE_HSPI_PORT

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible
// then uncomment "use hspi".
                                                                 
// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!
// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS
```
All the best! 👍

closed 2023-07-12 by ctc-nick