Bodmer/JPEGDecoder #82: Compilation problem with ESP8266

Ludovic-zibelli, 2023-08-15:
Hello, I have this defect when I want to compile with the arduino IDE I put the errors can you help me ?

```
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:91:3: error: 'File' does not name a type; did you mean 'SdFile'?
   91 |   File g_pInFileSd;
      |   ^~~~
      |   SdFile
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:144:21: error: 'File' has not been declared
  144 |   int decodeSdFile (File g_pInFile);
      |                     ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'uint8_t JPEGDecoder::pjpeg_need_bytes_callback(uint8_t*, uint8_t, uint8_t*, void*)':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:84:34: error: 'g_pInFileSd' was not declared in this scope; did you mean 'g_pInFileFs'?
   84 |  if (jpg_source == JPEG_SD_FILE) g_pInFileSd.read(pBuf,n); // else we are handling a file
      |                                  ^~~~~~~~~~~
      |                                  g_pInFileFs
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'int JPEGDecoder::decodeSdFile(const char*)':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:353:2: error: 'File' was not declared in this scope; did you mean 'fs::File'?
  353 |  File pInFile = SD.open( pFilename, FILE_READ);
      |  ^~~~
      |  fs::File
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:50,
                 from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
C:\Users\ludov\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/FS.h:52:7: note: 'fs::File' declared here
   52 | class File : public Stream
      |       ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:355:22: error: 'pInFile' was not declared in this scope
  355 |  return decodeSdFile(pInFile);
      |                      ^~~~~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'int JPEGDecoder::decodeSdFile(const String&)':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:362:2: error: 'File' was not declared in this scope; did you mean 'fs::File'?
  362 |  File pInFile = SD.open( pFilename.c_str(), FILE_READ);
      |  ^~~~
      |  fs::File
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:50,
                 from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
C:\Users\ludov\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/FS.h:52:7: note: 'fs::File' declared here
   52 | class File : public Stream
      |       ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:364:22: error: 'pInFile' was not declared in this scope
  364 |  return decodeSdFile(pInFile);
      |                      ^~~~~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: At global scope:
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:371:5: error: 'int JPEGDecoder::decodeSdFile' is not a static data member of 'class JPEGDecoder'
  371 | int JPEGDecoder::decodeSdFile(File jpgFile) { // This is for the SD library
      |     ^~~~~~~~~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:371:31: error: 'File' was not declared in this scope; did you mean 'fs::File'?
  371 | int JPEGDecoder::decodeSdFile(File jpgFile) { // This is for the SD library
      |                               ^~~~
      |                               fs::File
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:50,
                 from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
C:\Users\ludov\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/FS.h:52:7: note: 'fs::File' declared here
   52 | class File : public Stream
      |       ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'void JPEGDecoder::abort()':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:473:38: error: 'g_pInFileSd' was not declared in this scope; did you mean 'g_pInFileFs'?
  473 |  if (jpg_source == JPEG_SD_FILE) if (g_pInFileSd) g_pInFileSd.close();
      |                                      ^~~~~~~~~~~
      |                                      g_pInFileFs

exit status 1

Compilation error: exit status 1

```

Thank you

Bodmer, 2023-08-15:
It looks like there is no SD card library in your software environment.

So comment out this line:
https://github.com/Bodmer/JPEGDecoder/blob/master/src/User_Config.h#L5

Ludovic-zibelli, 2023-08-17:
Hello, I can't find where the problem comes from, I still have the same problem. I forgot to specify that I have the latest version of the Arduino IDE

Ludovic-zibelli, 2023-09-24:
Hello, Help

In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:91:3: error: 'File' does not name a type; did you mean 'SdFile'?
   91 |   File g_pInFileSd;
      |   ^~~~
      |   SdFile
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:144:21: error: 'File' has not been declared
  144 |   int decodeSdFile (File g_pInFile);
      |                     ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'uint8_t JPEGDecoder::pjpeg_need_bytes_callback(uint8_t*, uint8_t, uint8_t*, void*)':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:84:34: error: 'g_pInFileSd' was not declared in this scope; did you mean 'g_pInFileFs'?
   84 |  if (jpg_source == JPEG_SD_FILE) g_pInFileSd.read(pBuf,n); // else we are handling a file
      |                                  ^~~~~~~~~~~
      |                                  g_pInFileFs
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'int JPEGDecoder::decodeSdFile(const char*)':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:353:2: error: 'File' was not declared in this scope; did you mean 'fs::File'?
  353 |  File pInFile = SD.open( pFilename, FILE_READ);
      |  ^~~~
      |  fs::File
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:50,
                 from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
C:\Users\ludov\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/FS.h:52:7: note: 'fs::File' declared here
   52 | class File : public Stream
      |       ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:355:22: error: 'pInFile' was not declared in this scope
  355 |  return decodeSdFile(pInFile);
      |                      ^~~~~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'int JPEGDecoder::decodeSdFile(const String&)':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:362:2: error: 'File' was not declared in this scope; did you mean 'fs::File'?
  362 |  File pInFile = SD.open( pFilename.c_str(), FILE_READ);
      |  ^~~~
      |  fs::File
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:50,
                 from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
C:\Users\ludov\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/FS.h:52:7: note: 'fs::File' declared here
   52 | class File : public Stream
      |       ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:364:22: error: 'pInFile' was not declared in this scope
  364 |  return decodeSdFile(pInFile);
      |                      ^~~~~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: At global scope:
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:371:5: error: 'int JPEGDecoder::decodeSdFile' is not a static data member of 'class JPEGDecoder'
  371 | int JPEGDecoder::decodeSdFile(File jpgFile) { // This is for the SD library
      |     ^~~~~~~~~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:371:31: error: 'File' was not declared in this scope; did you mean 'fs::File'?
  371 | int JPEGDecoder::decodeSdFile(File jpgFile) { // This is for the SD library
      |                               ^~~~
      |                               fs::File
In file included from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.h:50,
                 from c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:39:
C:\Users\ludov\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/FS.h:52:7: note: 'fs::File' declared here
   52 | class File : public Stream
      |       ^~~~
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp: In member function 'void JPEGDecoder::abort()':
c:\Users\ludov\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src\JPEGDecoder.cpp:473:38: error: 'g_pInFileSd' was not declared in this scope; did you mean 'g_pInFileFs'?
  473 |  if (jpg_source == JPEG_SD_FILE) if (g_pInFileSd) g_pInFileSd.close();
      |                                      ^~~~~~~~~~~
      |                                      g_pInFileFs
Plusieurs bibliothèque trouvées pour "TFT_eSPI.h"
  Utilisé: C:\Users\ludov\OneDrive\Documents\Arduino\libraries\TFT_eSPI
  Non utilisé: C:\Users\ludov\OneDrive\Documents\Arduino\libraries\TFT_eSPI-master
exit status 1

Compilation error: exit status 1

Bodmer, 2023-09-24:
It looks like the newer board support package no longer defines ESP8266 but instead defined ARDUINO_ARCH_ESP8266

I have updated the library so it should compile fine now.

closed 2023-09-24 by Bodmer

Ludovic-zibelli, 2023-09-24:
Thanks it works now