Bodmer/JPEGDecoder #84: esp_littlefs: mount failed

NhanDoremon, 2023-12-27:
I am trying to follow this tutorial to show Jped Image wih JPEGDecoder: https://www.youtube.com/watch?v=4eD6recBq4M
I get the following error:

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
File Test!
INIT!
E (941) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1225:error: Corrupted dir pair at {0x0, 0x1}
E (941) esp_littlefs: mount failed,  (-84)
E (944) esp_littlefs: Failed to initialize LittleFS

Can someone please tell me what I need to do to fix this?

NhanDoremon, 2023-12-27:
Here is all my code, I use 320x480 Tft LCD and ESP32 DevKit1:
#include "SPIFFS.h"

// JPEG decoder library
#include <JPEGDecoder.h>
#include <SPI.h>
#include "Adafruit_ST7796S_kbv.h"
#include "Adafruit_GFX.h"

#define TFT_MISO    19
#define TFT_MOSI    23
#define TFT_SCLK    18
#define TFT_BL      32       // TFT backlight  pin
#define TFT_BACKLIGHT_ON HIGH

// Define pin CS CD and RST for Each LCD
#define TFT_CS      13      // Chip select control pin
#define TFT_RST     12
#define TFT_DC      14      // Data Command control pin

//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);
//Adafruit_ST7796S_kbv tft(TFT_CS,TFT_DC, TFT_RST);
Adafruit_ST7796S_kbv tft = Adafruit_ST7796S_kbv(TFT_CS, TFT_DC, TFT_RST);
boolean SPIFFSInited = false;
int imageIndex = 0;

void setup(void) {
  
//  long unsigned debug_start = millis ();
//  while (!Serial && ((millis () - debug_start) <= 5000)) ;
  Serial.begin(115200);
  Serial.println("File Test!"); 
  tft.begin();
  tft.setRotation(3);
//  tft.fillScreen(ILI9341_RED);
  tft.fillScreen(ST7796S_RED);

  Serial.println("INIT!");

  if (!SPIFFS.begin()) {
      Serial.println("SPIFFS initialisation failed!");
      while (1) yield(); // Stay here twiddling thumbs waiting
    }
    SPIFFSInited = true;
    Serial.println("SPIFFS initialisation success!");
}

void loop(){
  if(SPIFFSInited){

    if(imageIndex > 45){
      imageIndex = 0;
      Serial.println(" if(imageIndex > 45)");
    }

    String fileIndex = "/0";
    if(imageIndex < 10){
      fileIndex += "0" + String(imageIndex);
      Serial.println(" if(imageIndex > 10)");
    }else{
      fileIndex += "" + String(imageIndex);
      Serial.println(" if(imageIndex < 10)");
    }

    fileIndex += ".jpg";
    Serial.println("fileIndex : " + fileIndex);
    drawFSJpeg(fileIndex.c_str(), 0, 0);
    
    imageIndex++;
  }
[... 162 lines cut ...]
  Serial.println("  #include <avr/pgmspace.h>");
  Serial.println("#endif");
  Serial.println("");
  Serial.print("const uint8_t ");
  while (*filename != '.') Serial.print(*filename++);
  Serial.println("[] PROGMEM = {"); // PROGMEM added for AVR processors, it is ignored by Due

  while ( jpgFile.available()) {

    data = jpgFile.read();
    Serial.print("0x"); if (abs(data) < 16) Serial.print("0");
    Serial.print(data, HEX); Serial.print(",");// Add value and comma
    line_len++;
    if ( line_len >= 32) {
      line_len = 0;
      Serial.println();
    }

  }

  Serial.println("};\r\n");
  // jpgFile.seek( 0, SeekEnd);
  jpgFile.close();
}
////====================================================================================

NhanDoremon, 2023-12-28:
Thank you for all to see, I have sovled this problem. I think my ESP32 and another things maybe have conflict with LittleFS. So my solution is that: Replacing all the keyword of "LittleFS" to "SPIFFS in the main code. Example:  fs::File jpgFile = LittleFS.open( filename, "r");   ----> fs::File jpgFile = SPIFFS.open( filename, "r");
Moverover, I aslo changed all the command like above in JPEGDecoder.cpp and hide the command #include LittleFS.h.
So, that all my solution, after alterning it work well, but the speed to read and decoder Jpeg image is few slow. If you have any other solution. Please give a comment below for everyone.

Bodmer, 2023-12-28:
SPIFFS is being deprecated. I suggest that LittleFS is used.

RDJunc, 2024-10-16:
Bodmer, your libraries have been invaluable to a great many people, thanks very much.

I also ran across this problem and understand that LittleFS is the preferred file system.  All good but how to make that conversion is not very easy to find.  Might I suggest that you provide a description?

I changed all the SPIFFS reference to LittleFS but that still did not work.  I went into the JPEGDecoder.h file and commented out the #define SPIFFS LittleFS line.  I have my test code working but I am not sure this is the right way to resolve this issue.  Is this all that is needed?