Bodmer/JPEGDecoder #89: LittleFS declaration conflict error for ESP32S3

perigalacticon, 2024-11-11:
Arduino IDE version: 1.8.19 (latest).
ESP Boards version: 3.0.5 (latest).
Board selection: ESP32S3_Dev_Module.
Board Partition Scheme: Default 4MB with spiffs (1.2MB APP / 1.5MB SPIFFS).
Example ino file:  ESP32_SDcard_jpeg

When I compile this for ESP32S3 Dev Module I get this error:

`In file included from C:\Users\PERIGA~1\AppData\Local\Temp\arduino_modified_sketch_142262\ESP32_SDcard_jpeg.ino:16:
C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src/JPEGDecoder.h:48:20: error: conflicting declaration 'fs::SPIFFSFS LittleFS'
   48 |     #define SPIFFS LittleFS
      |                    ^~~~~~~~
C:\Users\perigalacticon\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\SPIFFS\src/SPIFFS.h:37:21: note: in expansion of macro 'SPIFFS'
   37 | extern fs::SPIFFSFS SPIFFS;
      |                     ^~~~~~
In file included from C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\JPEGDecoder\src/JPEGDecoder.h:47:
C:\Users\perigalacticon\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\LittleFS\src/LittleFS.h:37:23: note: previous declaration as 'fs::LittleFSFS LittleFS'
   37 | extern fs::LittleFSFS LittleFS;
      |                       ^~~~~~~~
Multiple libraries were found for "LittleFS.h"
 Used: C:\Users\perigalacticon\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\LittleFS
 Not used: C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\LittleFS_mod
Multiple libraries were found for "SD.h"
 Used: C:\Users\perigalacticon\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
 Not used: C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\SD
exit status 1
Error compiling for board ESP32S3 Dev Module.
`

I can't quite figure it out, I am not a professional programmer, but I'd really like to use your libraries.  Can you find the  solution for this?

perigalacticon, 2024-11-11:
I forgot I had moved the location of the JPEGDecoder.h include statement to the top of the includes list to see if it would clear up a different error I got previously.  Putting JPEGDecoder.h at the bottom of the includes list and this stopped the LittleFS library error, but now I get the error I had first:  "invalid conversion from 'uint32_t (*)(JDEC*, uint8_t*, uint32_t)' {aka 'long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)'} to 'UINT (*)(JDEC*, BYTE*, UINT)' {aka 'unsigned int (*)(JDEC*, unsigned char*, unsigned int)'} [-fpermissive]":

```
C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\TFT_eFEX\TFT_eFEX.cpp: In function 'bool jpgDecode(jpg_file_decoder_t*, uint32_t (*)(JDEC*, uint8_t*, uint32_t))':
C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\TFT_eFEX\TFT_eFEX.cpp:1646:41: error: invalid conversion from 'uint32_t (*)(JDEC*, uint8_t*, uint32_t)' {aka 'long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)'} to 'UINT (*)(JDEC*, BYTE*, UINT)' {aka 'unsigned int (*)(JDEC*, unsigned char*, unsigned int)'} [-fpermissive]
 1646 |     JRESULT jres = jd_prepare(&decoder, reader, work, 3100, jpeg);
      |                                         ^~~~~~
      |                                         |
      |                                         uint32_t (*)(JDEC*, uint8_t*, uint32_t) {aka long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)}
In file included from C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\TFT_eFEX\TFT_eFEX.h:23,
                 from C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\TFT_eFEX\TFT_eFEX.cpp:9:
C:\Users\perigalacticon\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32s3/include/esp_rom/include/esp32s3/rom/tjpgd.h:89:29: note:   initializing argument 2 of 'JRESULT jd_prepare(JDEC*, UINT (*)(JDEC*, BYTE*, UINT), void*, UINT, void*)'
   89 | JRESULT jd_prepare (JDEC *, UINT(*)(JDEC *, BYTE *, UINT), void *, UINT, void *);
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\TFT_eFEX\TFT_eFEX.cpp:1666:32: error: invalid conversion from 'uint32_t (*)(JDEC*, void*, JRECT*)' {aka 'long unsigned int (*)(JDEC*, void*, JRECT*)'} to 'UINT (*)(JDEC*, void*, JRECT*)' {aka 'unsigned int (*)(JDEC*, void*, JRECT*)'} [-fpermissive]
 1666 |     jres = jd_decomp(&decoder, jpgWrite, (uint8_t)jpeg->scale);
      |                                ^~~~~~~~
      |                                |
      |                                uint32_t (*)(JDEC*, void*, JRECT*) {aka long unsigned int (*)(JDEC*, void*, JRECT*)}
C:\Users\perigalacticon\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32s3/include/esp_rom/include/esp32s3/rom/tjpgd.h:90:28: note:   initializing argument 2 of 'JRESULT jd_decomp(JDEC*, UINT (*)(JDEC*, void*, JRECT*), BYTE)'
   90 | JRESULT jd_decomp (JDEC *, UINT(*)(JDEC *, void *, JRECT *), BYTE);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "SD.h"
 Used: C:\Users\perigalacticon\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
 Not used: C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\SD
Multiple libraries were found for "LittleFS.h"
 Used: C:\Users\perigalacticon\AppData\Local\arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\LittleFS
 Not used: C:\Users\perigalacticon\OneDrive\Documents\Arduino\libraries\LittleFS_mod
exit status 1
Error compiling for board ESP32S3 Dev Module.
```

Something is trying to convert from a 32bit var to an 8bit var.

HoaxParagon, 2025-02-06:
I'm also having compilation errors related to this library and multiple defines of SPIFFS and LittleFS. I haven't tried moving the library include around but I'm trying to get ESP32 Marauder software to compile and it won't because of this library. I could mod the library but I feel there's something else going on here. I'm trying to compile for the T-display S3 pro.

RootDmytro, 2025-04-03:
If anyone is still interested, I have created a minimal PR to fix this
[PR90](https://github.com/Bodmer/JPEGDecoder/pull/90)
I'm not sure if it's the best way to fix this, history of related changes and their logic is not entirely clear to me.
But it works, and looks logical.